mcu-debug/rtos-views #4: ChibiOS will detection can fail when it shouldnt

haneefdm, 2022-10-15:
There is an issue with the code below
```typescript
      try {
          this.chReglist = await this.getVarIfEmpty(this.chReglist, useFrameId, '(uint32_t) &ch_system.reglist', false);
          this.smp = true;
      } catch (e) {
          this.chReglist = await this.getVarIfEmpty(this.chReglist, useFrameId, '(uint32_t) &ch0.reglist', false);
      }
```

In the catch, you should check to see if the failure was because of a BusyError and if so, re-throw it. Assume failures when trying to get variables of expressions.

FYI: user could hit continue, step, or debugger itself can continue while we are in the middle of querying. Hence all the complications and care while evaluating expressions. Because of runToEntryPoint, this is guaranteed to happen and if not, you got lucky. Note also, different debuggers work differently on failure/busy conditions and I had to figure that out.

After detection, any evaluation can fail because of the same issue. Fast single stepping or user setting a breakpoint while the program is running and many other situations. We always expect to fail in the middle of something, but we will recover next time the program stops -- also why I print when data was collected and continue to show last valid data. I could have chosen to blank out the RTOS view, but I chose to show last valid results

Cc: @vrepetenko

vrepetenko, 2022-10-15:
could you please point me to this BusyError declaration or using example?

haneefdm, 2022-10-15:
Example is right there a few lines below the code I typed...You check for the type of the exception `e`

vrepetenko, 2022-10-15:
src/rtos/rtos-chibios.ts:131:38 - error TS2304: Cannot find name 'BusyError'.

131                     if (e instanceof BusyError) {

haneefdm, 2022-10-15:
It is not called BusyError -- instead actual name is SouldRetry See example just below where you are editing

https://github.com/mcu-debug/rtos-views/blob/7c34554840b6c5b5acd8629d81081fc92db6b820/src/rtos/rtos-ucosii.ts#L100

vrepetenko, 2022-10-15:
created PR #5

closed 2022-10-15 by haneefdm