mcu-debug/rtos-views #8: Zephyr implementation open points

PhilippHaefele, 2022-11-26:
- [ ] Inform user when CONFIG_THREAD_MONITOR is not set
- [ ] Check if stack_info.size is always in Bytes and not number of stack entries and adapt code if the size is in stack entries
- [ ] Add support for more arches and inform user when not supported (it's not always callee_saved.psp)
- [ ] Try to get information like real address and type of kernel objects (semaphore, mutex etc.) the tasks are pending on (we only have their _wait_q_t *pended_on right now) => when CONFIG_TRACING and CONFIG_TRACING_OBJECT_TRACKING are enabled, it seems that we have lists of all object types => see tracing_tracking.c for link list entry points

rosterloh, 2023-01-13:
Works for me 
[image]

hongshui3000, 2023-03-09:
very much looking forward to

PhilippHaefele, 2023-03-09:
> very much looking forward to

Anything in mind that you need, as in general Zephyr works?

hongshui3000, 2023-03-10:
All Zephyr objects supported by Zephyr's object tracking system (timers, memory slabs, semaphores, mutexes, stacks, message queues, mailboxes, pipes, and queues) are supported in the rtos-views

dbongartz, 2023-09-21:
Do understand correctly that the thread info for zephyr shows "current stack usage"?

If I use "CONFIG_THREAD_ANALYZER=y" I can see that this prints "maximum stack usage". Microsoft's "Embedded Tools" also print maximum size (but not current size).

I had a look at the code here and guess it could easily be added to show:

% Current / % Max (Current B / Max B / Size B)

haneefdm, 2023-09-21:
Thanks, @dbongartz Maybe one of our Zephyr experts will chime in. This is 99.8% user-contributed stuff.

PhilippHaefele, 2024-01-07:
@dbongartz

First of all, sorry for not responding. Need to find some spare time to work on the extension again.

> Do understand correctly that the thread info for zephyr shows "current stack usage"?

I just did check the code and i think it shows the "current" (last saved. See #28) stack usage. I should state that in the header somehow like i did for uC/OS-II.

stemschmidt, 2024-03-16:
This is such a great tool! Thank you!

axelnxp, 2024-08-20:
Hi,

Could we add the max stack usage to the list ?