mcu-debug/rtos-views #10: Project format settings

PhilippHaefele, 2022-11-27:
If i currently save a file it's formatted as intended via the settings.

But it seems that the settings that are used at my place are not used @haneefdm place.

I did find following issues so far:

- [ ] `"esbenp.prettier-vscode"` entry is missing in extensions.json
- [ ] because we have a `.editorconfig`, `"prettier.` settings in `settings.json` have no affect
- [ ] `indent_size` and `quote_type` settings are missing in `.editorconfig`
- [ ] trailing commas are added => seems to be ok, but do we want them?
- [ ] inner parentheses are removed in if statements (and dome other places) => i really like to add the inner parentheses for better readability when there is more than 1 condition part inside

I created https://github.com/mcu-debug/rtos-views/pull/9 for the first 3 points

haneefdm, 2022-11-27:
> railing commas are added => seems to be ok, but do we want them?

Where do you see this? So, the default is 'es5' it appears. I don't want trailing commas, so I added this. Will that suffice?
```
"prettier.trailingComma": "none",
```

> inner parentheses are removed in if statements (and dome other places) => i really like to add the inner parentheses for better readability when there is more than 1 condition part inside

This has been annoying me. This seems to be an open issue with prettierl

https://github.com/prettier/prettier/issues/187
https://github.com/prettier/prettier/issues/3805

And, I am never sure it did it right.

haneefdm, 2022-11-27:
I found this. Not sure this is exactly what we want.

https://github.com/kevin940726/eslint-plugin-no-mixed-operators

akhba01, 2022-12-08:
can someone help me
I write js code like this :
((HourNow / 12) * 360) + ((minuteNow / 60) * 360) + 90
but after I saved and prettier formatted it to: 
(HourNow / 12) * 360 + (minuteNow / 60) * 360 + 90;

PhilippHaefele, 2023-01-09:
I need to check this tomorrow, but it seems that we can use eslint for formatting only. See https://daveceddia.com/vscode-use-eslintrc/ 

We could get rid of prettier then and most likely the eslint plugin from above will work also.

@haneefdm What do you think about this approach?

haneefdm, 2023-02-20:
I am closing this for now.

closed 2023-02-20 by haneefdm