mcu-debug/rtos-views #14: Fail to detect FreeRTOS with SMP enabled

howard0su, 2023-01-14:
This is because pxCurrentTCB is renamed to pxCurrentTCBs in order to support multi cores.
>>> p pxCurrentTCB
No symbol "pxCurrentTCB" in current context.
>>> p pxCurrentTCBs
$3 = {[0] = 0x20016598 <ucHeap+39100>, [1] = 0x20016100 <ucHeap+37924>}

I verified other symbols are not changed.

haneefdm, 2023-01-14:
Thank you for notifying us this use case. I need to create a test case for this. Some questions please

1. May I enquire which multi-core device you are using?
2. What are some of the rules to follow for enabling SMP? I do have 1 dual-core board but with separate memories, caches and some combined memories. Wasn't sure how the stack(s) worked or how the scheduler works. I was assuming they would have to either be using common memory or use some sort of thread affinity to specify which thread should run on which processor. Maybe I am overcomplicating things
3. What is a good way to determine the number of cores SMP is enabled for? Say I have 4 cores but I only want to use the RTOS on 3 of those cores.
4. Also, can I enable SMP on a single-core processor, and what might be the consequences?

I am asking these questions to get some knowledge upfront for developing and testing for SMP. Saves me time and accelerates SMP support. I don't want to spend months figuring it out when experts already have answers.

What other gotchas I gotta worry about?

howard0su, 2023-01-15:
> Thank you for notifying us this use case. I need to create a test case for this. Some questions please
> 
> 1. May I enquire which multi-core device you are using?
I am using RPI Pico, which is one of official FreeRTOS supported SMP platform

> 2. What are some of the rules to follow for enabling SMP? I do have 1 dual-core board but with separate memories, caches and some combined memories. Wasn't sure how the stack(s) worked or how the [...]
SMP only supports that two cores are sharing the same memory (maybe with different CPU cache) Stacks are allocated in the same memory region.

> 3. What is a good way to determine the number of cores SMP is enabled for? Say I have 4 cores but I only want to use the RTOS on 3 of those cores.
There are config to config how many cores to use. and Tasks can set core affinity. I don't think FreeRTOS supports the scenarios you mentioned. 
> 4. Also, can I enable SMP on a single-core processor, and what might be the consequences?
I didn't read through code. I think it should be fine.
> 
> I am asking these questions to get some knowledge upfront for developing and testing for SMP. Saves me time and accelerates SMP support. I don't want to spend months figuring it out when experts already have answers.
Thanks. My suggestion is using PC or Pico to try and step through. I am not FreeRTOS expert and happened to use it for my current project with Pico.
> 
> What other gotchas I gotta worry about?

howard0su, 2023-01-15:
Also, this is a PR I sumitted for Freertos_gdb python, which is same issue. https://github.com/espressif/freertos-gdb/pull/3

bdmihai, 2023-02-21:
Adding the missing symbol to freertos port seems to work... 

.global pxCurrentTCB
pxCurrentTCB:
.word pxCurrentTCBs

PhilippHaefele, 2023-02-21:
I don't think just using `pxCurrentTCBs` or adding `pxCurrentTCB` is the right solution, as like @haneefdm mentioned it depends what `pxCurrentTCBs` really is at the end. This most likely also depends on configuration options etc.

I just looked into the code and found following infos:

`PRIVILEGED_DATA TCB_t * volatile pxCurrentTCBs[ configNUM_CORES ] = { NULL };` from https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/smp/tasks.c#L329

So it seems that the SMP version of FreeRTOS (separate branch right now) also does support single core usage (at least for the RP2040. See following comment and corresponding code: 
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/smp/portable/ThirdParty/GCC/RP2040/port.c#L47 

For this special case (using SMP with only one core) the above mentioned fixes work. Which most likely is not really the intension of the SMP version.

For the case more cores are used (configNUM_CORE > 1) there's most likely more to check => https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/smp/tasks.c

haneefdm, 2023-02-21:
@PhilippHaefele You beat me to it. Thanks.

When we do an SMP, we will only seriously do it for a real use case of actually multiple cores

hwmland, 2023-12-21:
@haneefdm Do you have any plans to add this SMP support?

bdmihai, 2023-12-21:
Supporting SMP would be nice... Starting with v11.0.0 FreeRTOS has-it on mainline.

PR #44 (2025-01-25): Add support for SMP on FreeRTOS
This PR resolves #14  and resolves #43
closed 2025-01-25 by haneefdm