mcu-debug/rtos-views #15: how to install

venu942000, 2023-01-25:
iam working with stm32 micro controller and aslo iam using cortex debug but in your readme there is installation is given

please provide me any file for installation guide

haneefdm, 2023-01-26:
Not sure what you mean. You install it from the VSCode marketplace or right from within VSCode extensions section where you manage all extensions. Like every other extension.

https://marketplace.visualstudio.com/items?itemName=mcu-debug.rtos-views

Also, if you install the latest version of Cortex-Debug, it will auto-install it for you ad this is a dependency.

Can you explain a bit more of what is going on?

haneefdm, 2023-01-26:
Please close this issue if you are doing okay

venu942000, 2023-01-26:
iam new to vs code 

iam using stm32f4 and i configure everything  and work fine with cotex debug now i need to trace the rtos memory usage so i use rtos view .

when i start the debug , in xrtos tab the view is like ?????? no data is preview . can you understand my problem

venu942000, 2023-01-26:
i have installed rtos views directly on vs code

haneefdm, 2023-01-26:
Nope, I don't understand. You have not said which RTOS -- every RTOS is totally different. We sometimes give you hints to say how to get more info. There are dozens of #defines for a given RTOS you have to enable to see all the info.

How about a screenshot (not your whole screen, just the RTOS panel part).

venu942000, 2023-01-27:
hi

i will send the screenshot today

venu942000, 2023-01-27:
this happens when debugging the controller []() 
[image]

haneefdm, 2023-01-27:
Did you notice on the bottom instructions on getting the missing fields?

haneefdm, 2023-01-27:
You disable all that info in your FW.

venu942000, 2023-01-27:
> You disable all that info in your FW.

is there any sample or example

haneefdm, 2023-01-27:
Did you scroll down or make the window bigger? Instructions should be right there. Expand the hints. Also, read the FreeRTOS docs and header file for configuration.

[image]

If the hint is missing we have a different problem.

venu942000, 2023-01-27:
sorry there is no hint is there . ok i will check my firmware but i dont see any hint in my window

haneefdm, 2023-01-27:
Unless you can provide a test case where I can duplicate the issue, not much I can do. Here is what I see when info is missing. Of course you have to click on the Blue text to see the full text.

[image]