mcu-debug/rtos-views #17: hint is not visible in xrtos window

venu942000, 2023-01-27:


haneefdm, 2023-01-27:
Need test case. See screenshot in the following issue. There is zero probability that I can reproduce what you have.

https://github.com/mcu-debug/rtos-views/issues/15#issuecomment-1406663697

haneefdm, 2023-01-27:
You also did not attach a full screenshot of the window. I saw a partial one, I want to see the full screenshot.

closed 2023-01-28 by venu942000

xy-23, 2024-07-13:
Hi, I also met this problem. I've tried both on macOS Sonoma 14.5 and Windows on Arm 11
[image]

xy-23, 2024-07-13:
It seems related to the thread id. As the following images shows, if the thread id disappears, so does the hint. However if the thread id behaves normal, then the hint can display.
[image]
[image]

haneefdm, 2024-07-13:
Hmmmm. Thanks for reporting this.