mcu-debug/rtos-views #18: The Runtime statistics are not visible with FreeRTOS

MedvedevDavid, 2023-01-29:
Even after the configGENERATE_RUN_TIME_STATS set to 1 and portCONFIGURE_TIMER_FOR_RUN_TIME_STATS() and portGET_RUN_TIME_COUNTER_VALUE() are defined the runtime data is showing question marks in the FreeRTOS with stm32 F401.

haneefdm, 2023-01-29:
Getting Runtime is pretty complicated and it took me some time to get everything right.

Btw, did you try Microsoft Embedded Tools and see if you can get the same? We both support FreeRTOS.

Since you have those stats not showing, do you see a hint (please scroll all the way down), and perhaps our instructions are missing something? I have a feeling some #define is missing or wrong

haneefdm, 2023-01-29:
Using the normal debugger, examine/watch the thread data structures to see if the field `ulRunTimeCounter` exists or not. If that is missing, then we have no info. See what combo of #defines you have to mess with to get that field in the thread information.

Same applies to the global `ulTotalRunTime` and make sure that is defined. Or else, we cannot compute a %age

MedvedevDavid, 2023-01-30:
> Using the normal debugger, examine/watch the thread data structures to see if the field `ulRunTimeCounter` exists or not. If that is missing, then we have no info. See what combo of #defines you [...]

Thank you haneefdm, the issue was that the ulTaskSwitchedInTime and ulTotalRunTime was optimized making them to volatile solved the issue.

closed 2023-01-30 by haneefdm