mcu-debug/rtos-views #19: threadx supported?

frank-amy, 2023-02-01:
Hello,  do we have some plans about supporting Threadx

haneefdm, 2023-02-01:
Not unless we can find a ThreadX expert to contribute. If you see our readme, most support is from contributors. I own the extension but I cannot be an expert at every RTOS :-)

Btw, MS Embedded tools supports RTOS views and we complement each other. Install their extension. They have support for cortex-debug. Not sure which debugger you are using

https://marketplace.visualstudio.com/items?itemName=ms-vscode.vscode-embedded-tools

We may not have feature parity but we display some of the same info. Good luck

Please close this issue unless you want to contribute., It is a lot of work, I warn you.

frank-amy, 2023-10-31:
thanks very much, I've tried MS embedded tool, bus it still has some something wrong to show the usage

closed 2023-10-31 by frank-amy