mcu-debug/rtos-views #20: embOS detected but unable to collect RTOS information

NOhs, 2023-02-15:
Hello there,

> This is a duplication of https://github.com/Marus/cortex-debug/issues/826, which was posted in the wrong issue tracker by me. 

## Issue Description
Currently, I have embOS (5.18.0.0) running on a little evaluation board RT600 (with an Cortex M33 chip) and I can use embOS-View (5.18.0.2) to see the different tasks and their stacks etc. If I try to use the XRTOS window of **cortex-debug**, I get the message (while pausing at a break point):

> **RTOS Views: Session Name: "armclang", embOS detected.**
> Unable to collect full RTOS information. Could not read any task from "OS_Global.pTask". Perhaps program is busy or did not stop long enough

If I use the default variable watch feature of **cortex-debug**, I CAN access `OS_Global.pTask`. So I am a bit confused as to what is going on. Do I need a special configuration for **cortex-debug** to handle embOS? I currently only have:

```json
"rtos": "embOS",
```
in my launch.json for **cortex-debug**.

## System Details

- **Board:** [RT600 evaluation board](https://www.nxp.com/design/development-boards/i-mx-evaluation-and-development-boards/i-mx-rt600-evaluation-kit:MIMXRT685-EVK)
- **embOS Version**: 5.18.0.0
- **embOS View**: 5.18.0.2
- **Host System**: Windows
- **Compiler**: Armclang 6.19
- **JLink / JLink GDB Server**: 7.50
- **ARM Toolchain**: GNU Arm embedded 10 2021.10

PhilippHaefele, 2023-02-16:
Thanks for reporting this here again.

I think i just found the error. I will test a little bit more to make sure everything works as expected.

haneefdm, 2023-02-16:
Thanks, @PhilippHaefele.

I/We need to create a template for bug submission. Not just this one but for all of our repos.

PhilippHaefele, 2023-02-16:
> I/We need to create a template for bug submission. Not just this one but for all of our repos.

I was also thinking this the other day. Do you know of any good template we can adjust to our needs?

haneefdm, 2023-02-16:
We could use the cpptools as a template. They ask mostly the same questions, except we have some specific things gdb-server, etc.

https://github.com/microsoft/vscode-cpptools/tree/main/.github

Let me know if you see something better.

haneefdm, 2023-02-16:
Created issue https://github.com/mcu-debug/rtos-views/issues/21

commit 60505c92 (2023-02-20): Fixing object change error in embOS (fixes #20)
commit f4efb80e (2023-02-26): Do show timeouts in a right manner
Fixes #20
closed 2023-02-20 by haneefdm