mcu-debug/rtos-views #37: [FR] Add support for Keil RTX

augustozanellato, 2023-03-06:
Hi, 
it would be nice to have support for Keil's RTX RTOS, it's used in MBED OS which has a considerable user base.
Best Regards,
Augusto

haneefdm, 2023-03-06:
Maybe will find a volunteer. I know nothing of RTX. To do this, you have to know the internals of the RTOS and available options. It is not just something we can do quickly or on a whim.

Maybe @thegecko could find a volunteer for us.

augustozanellato, 2023-03-06:
> Maybe will find a volunteer. I know nothing of RTX. To do this, you have to know the internals of the RTOS and available options. It is not just something we can do quickly or on a whim.

No worries, I imagine it is non trivial to do. I just logged an issue here in the hope someone knowledgeable about it volunteers its time to implement it.

thegecko, 2023-03-07:
@jkrech Keil RTX support would be a great addition to this library, what do you think?

jkrech, 2023-03-07:
The essential knowledge should be all in here:
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/RTX5.scvd
The SCVD format is explained here: https://arm-software.github.io/CMSIS-View/main/SCVD_Format.html 
Not sure how easy it is to port.

silvio3105, 2023-10-05:
It would be nice to have RTX5.
I'm not sure how to add it, even with .scvd files. How this extension even works?

thorstendb-ARM, 2025-09-17:
Added support for Arm/RTX5 Threads.
PR https://github.com/mcu-debug/rtos-views/pull/76/