mcu-debug/rtos-views #38: Is it possible to use the live watch method to monitor rtos

hongshui3000, 2023-03-11:
Is it possible to use the live watch method to monitor rtos？

haneefdm, 2023-03-11:
Yes and no. It takes more than a second each time we query for an RTOS because we do it with many GDB transactions. This is why we print how long it took for the last query so you know how expensive it is.

But maybe update once a second or two perhaps You may have to disable certain things like looking for stack peaks which makes it even less useful.

You should use a professional/commercial tool that can monitor events either via instrumentation or via the trace port. They have proprietary HW, FW and SW to do all of that.

haneefdm, 2023-03-11:
We can bypass gdb and do raw reads and reconstruct data structures in our extension. We can probably gin 70% runtime. It will be very error prone and we have to do it for so many RTOSes.

If I think it will give me an order-of-magnitude performance, I would do that. But I don't think I will get that. The RTOS data structures are scattered all over your memory

hongshui3000, 2023-03-11:
Maybe something like RTT trace（a professional/commercial tool） is better instead of live watch, if live watch is so slow

closed 2023-03-11 by hongshui3000

haneefdm, 2023-03-11:
Cortex-Debug RTT is just as fast as others. If you use a good HW probe like JLink. Cortex-Debug already supports RTT and sometimes better than even JLink if you use it with OpenOCD. You can even graph with it.

You are confusing live-watch with rtos-views. RTOS views are super expensive. We are talking of looking at dozens of variables are data structures. Live-watch only looks at what you ask.

PhilippHaefele, 2023-03-11:
I personally use Segger SystemView when i want a detailed Live View of the RTOS.

You only need to keep in mind that i does influence the runtime (also depending on the granularity you do use trace for tracing).

haneefdm, 2023-03-11:
Yes, Segger, IAR, Lauterbach, uVision and even Percepio. Percepio is the odd one in this bunch because it requires instrumentation, but can run over RTT.

hongshui3000, 2023-03-11:
In fact, I tried Percepio, the only problem is that I can't use a debugger such as jlink at the same time as cortex debug