mcu-debug/rtos-views #41: Show FreeRTOS Queues

srobertjames, 2023-09-29:
**Is your feature request related to a problem? Please describe.**
When debugging a FreeRTOS project, I need to know the contents of a FreeRTOS Queue

**Describe the solution you'd like**
Similar to vscode watches: A table or tree showing FreeRTOS Queues, their size, and their contents (as an array)

**Describe alternatives you've considered**
1. Looking at the `u` member in the watches, and then looking at the memory view
2. Other debuggers seem to support this

srobertjames, 2023-10-31:
Are you able to respond to this?

PhilippHaefele, 2024-01-07:
@srobertjames Sorry that nobody responded yet. @haneefdm did the general and FreeRTOS implementation, but our free time is somehow limited.

Maybe someone else (e.g. also me) can help when there's spare time for this.

We're also happy to receive PullRequests.

So most likely for now this means that this feature will not be implemented very soon.

haneefdm, 2024-01-07:
I will take a look in a couple of weeks. Had been very busy lately.

PR #45 (2025-04-11): Show FreeRTOS Queues
Convert view to panels. Add panels for queues and semaphores/mutexes. Resolves #41 
Some example shots:
[image]
[image]
[image]
[image]
closed 2025-04-11 by haneefdm