mcu-debug/rtos-views #42: How can i view my task name?

USTHzhanglu, 2023-12-21:
[image]
I used freeRTOS in my mcu, the Task Name just show address, i don't now how to view it. if somebody know,please help me,thanks

USTHzhanglu, 2023-12-21:
by debug， i get this message `tmpThName："0x2000090c <ucHeap+1080> <error reading variable>"`

USTHzhanglu, 2023-12-21:
[image]
and i can read str by `MemoryView` like this.

This is some log in `rtos-debug`

```js
const result = await this.session.customRequest(cmd, arg);
/**
cmd is `evaluate`
arg is 
{
  frameId: 4096,
  expression: "(char *)((TCB_t*)0x200008d8).pcTaskName",
  context: "hover",
}

resault is

{
  result: "0x2000090c <ucHeap+1080> <error reading variable>",
  type: "char *",
  presentationHint: {
  },
  variablesReference: 3145788,
  memoryReference: "0x2000090c",
}

**/
```
hope this log helpful

PhilippHaefele, 2023-12-22:
Hi @USTHzhanglu 
Can you please state your system versions like VSCode, extension and most important FreeRTOS.

Without these we're not able to help you at all

USTHzhanglu, 2023-12-22:
> Hi @USTHzhanglu Can you please state your system versions like VSCode, extension and most important FreeRTOS.
> 
> Without these we're not able to help you at all

OS: Windows 11 22H2
Vscode: 1.85.1
extension : 0.0.7
freeRTOS: FreeRTOSv202112.00
probe: stlink
debuger: openocd  0.12.0-rc2-g7a1adfbec (2022-11-06-10:08)

htcrane, 2024-04-08:
I am facing a similar issue as well: only the first character of a task name will appear in the XRTOS table. When I use `char* test = pcTaskGetName(NULL);` to get the task name address, it appears as the full name with null termination in memory view.

OS: MacOS 14.4.1
Vscode: 1.88.0
Extension: 0.0.7
Cortex Debug Version: 1.12.1
FreeRTOS: 10.5.1
Probe: JLINK-EDU
Debugger: arm-none-eabi tool chain v13.2.1Rel1

thirtytwobits, 2024-06-26:
Been seeing this for months now so it probably showed up in an earlier vscode revision. Used to work with the same version of FreeRTOS under test:

Mac OS: 14.5 (M2)
vscode: 1.90.2 
mcu-debug RTOS Views: 0.0.7
FreeRTOS: 10.4.6
Probe: Jtrace
Debugger: gdb from 10.3

PhilippHaefele, 2024-08-18:
Hi together,

was able to test this today and can't see an issue with the task names.

I used:

Mac OS: 10.15.7 (Intel based Device)
VS Code: 1.92.2
mcu-debug RTOS Views: 0.0.7
FreeRTOS (Kernel): 11.1.0
Debug Plugin: Cortex Debug
Probe / GDB Server: Segger J-Link OB with SEGGER J-Link GDB Server V7.98d Command Line Version
GSB Client / Tools: gcc-arm-none-eabi-10.3-2021.10

Thinking of the cause of this. How did you create the task?
I did use `xTaskCreateStatic` for creating it.
According to @USTHzhanglu Debug output this seems to be some other method.
Will try to find some more time for testing and I'm happy for any feedback/testing from your side

USTHzhanglu, 2024-08-19:
> Hi together,
> 
> was able to test this today and can't see an issue with the task names.
> 
> I used:
> 
> Mac OS: 10.15.7 (Intel based Device) VS Code: 1.92.2 mcu-debug RTOS Views: 0.0.7 FreeRTOS (Kernel): 11.1.0 Debug Plugin: Cortex Debug Probe / GDB Server: Segger J-Link OB with SEGGER J-Link GDB Server V7.98d Command Line Version GSB Client / Tools: gcc-arm-none-eabi-10.3-2021.10
> 
> Thinking of the cause of this. How did you create the task? I did use `xTaskCreateStatic` for creating it. According to @USTHzhanglu Debug output this seems to be some other method. Will try to find some more time for testing and I'm happy for any feedback/testing from your side
> 
> Edit: Also tested `xTaskCreate` with Heap 1 implementation

like this

```c
  xReturn = xTaskCreate((TaskFunction_t )DebugTask,  /* 任务入口函数 */
                        (const char*    )"DebugTask",/* 任务名字 */
                        (uint16_t       )64,  /* 任务栈大小 */
                        (void*          )NULL,/* 任务入口函数参数 */
                        (UBaseType_t    )1, /* 任务的优先级 */
                        (TaskHandle_t*  )&DebugTask_Handle);/* 任务控制块指针 */  
```

USTHzhanglu, 2024-08-19:
> Hi together,
>
> was able to test this today and can't see an issue with the task names.
>
> I used:
>
> Mac OS: 10.15.7 (Intel based Device) VS Code: 1.92.2 mcu-debug RTOS Views: 0.0.7 FreeRTOS (Kernel): [...]

it's very bad that i can't reproduce this issue.:(
i change my computer and mcu chip.
i try my code with other chip and Prober, and they all show name as expected

PhilippHaefele, 2024-08-19:
> it's very bad that i can't reproduce this issue.:(
> 
> i change my computer and mcu chip.
> 
> i try my code with other chip and Prober, and they all show name as expected

What was the "old" MCU?

We also have other issues with task names not working and i thought that I saw a pattern there (Mac based on the Apple silicon / M series), but @USTHzhanglu did also see them on Windows 11.
So I currently do not see a good path to get further...

haneefdm, 2024-08-19:
My hunch is that I would focus on the version of gdb being used -- the requests and responses.

That is, all else being the same -- same FW (including RTOS) , gdb-server, cortex-debug, etc. I don't think the host OS or version of VSCode has much to do with it.

PhilippHaefele, 2024-09-14:
@haneefdm Was also targeting GDB and was thinking that there're maybe differences because they where compiled for a different target. As I said that was already proven wrong, as others also had issues.

@USTHzhanglu Did have a look once more. Just found that we use `char *` instead of `const char *` what is at least used in the FreeRTOS version i use (version & configuration).
As I'm not able to reproduce this at my side, can you please add the frist part `(TCB_t *)ADDRESS_OF_TASK_TCB` of the request as a WATCH expression, so we can see if at least this works and how `pcTaskName` shows up there.

The new `debug.showVariableTypes` / `"debug.showVariableTypes": true` may also help to get some more info.

Have added a screenshot where I did it:
[image]

Can you also provide me your `FreeRTOSConfig.h`. Maybe this also helps with testing

USTHzhanglu, 2024-09-23:
> What was the "old" MCU?
>
> We also have other issues with task names not working and i thought that I saw a pattern there (Mac based on the Apple silicon / M series), but @USTHzhanglu did also see [...]

GD32F310 .i have buy a new board, and will test after some time :)

USTHzhanglu, 2024-10-24:
> GD32F310 .i have buy a new board, and will test after some time :)

very sad. by this chip i also can`t reiteration. 
 If I see it again, I'll report in.

PhilippHaefele, 2026-01-20:
Closing as not reproducible and nobody stated other for quite some time now

closed 2026-01-20 by PhilippHaefele