mcu-debug/rtos-views #47: Why there is some missing info?

J-Rios, 2024-02-06:
Hello,

Really nice plugin to work with RTOS projects, thanks a lot for this.

I started to use this plugin with FreeRTOS v8 and have the same missing information (stack sized, stack used, stack peak, etc) as you have shown in one of the images on the README file:

[image]

I just want to ask, why is this information missing and if there is anything that I could do to configure the plugin or the project to get this info, or if it just a limitation depending on the type of RTOS and RTOS version used?

Best Regards.

haneefdm, 2024-02-06:
Because those are compile time options. Did you click on hints link at the bottom of the image?

PhilippHaefele, 2024-02-06:
Hi @J-Rios 

you need to click on `Hints to get more out of the FreeRTOS viewer` below the table to get information on how to get the missing values working.

haneefdm, 2024-02-06:
@J-Rios Please close this if it worked for you

J-Rios, 2024-02-07:
Oh, I did not realize that text was clickable, I will check...

closed 2024-02-07 by haneefdm