mcu-debug/rtos-views #49: xrtos view for FreeRTOS truncates task names

thirtytwobits, 2024-05-06:
Sometime in the last few months the xrtos view for FreeRtos started only displaying the first character of the task name:

[image]

I've confirmed that the task names still show normally using SEGGER Ozone with FreeRTOS.

thirtytwobits, 2024-05-09:
[image]

The contents of the cell is just the first character so this isn't just a UI clipping bug.

thirtytwobits, 2024-05-09:
Also note that the task names in the CALL STACK window are correct.

PhilippHaefele, 2024-06-16:
@thirtytwobits ease state your versions (VSCode, extension, FreeRTOS) and debug configuration.

I will try to reproduce the issue.

Issue was also mentioned in https://github.com/mcu-debug/rtos-views/issues/42#issuecomment-2041767784

thirtytwobits, 2024-06-26:
Been seeing this for months now so it probably showed up in an earlier vscode revision. Used to work with the same version of FreeRTOS under test:

Mac OS: 14.5 (M2)
vscode: 1.90.2
mcu-debug RTOS Views: 0.0.7
FreeRTOS: 10.4.6
Probe: Jtrace
Debugger: gdb from 10.3

xy-23, 2024-07-13:
Hi, I have the same issue.

macOS 14.5 (M1)
vscode 1.91.1
mcu-debug RTOS Views: 0.0.7
FreeRTOS CMSIS_V1

thirtytwobits, 2024-08-16:
Any updates on this? Still seeing this after upgrading to FreeRTOS 11

patrick-dojofive, 2024-09-05:
I'm seeing this issue as well.

macOS 14.6.1 (M2 Max)
vscode 1.92.1
mcu-debug 0.0.7
FreeRTOS 10.5.1

PhilippHaefele, 2024-09-21:
Can someone from this issue, do what i asked in the following comment https://github.com/mcu-debug/rtos-views/issues/42#issuecomment-2351129796 please?

As I suspect that the two issues are related to each other, we can maybe solve both at the same time.