mcu-debug/rtos-views #50: ChibiOS not detected

Woodgnome, 2024-05-22:
Using Cortex-Debug extension and ChibiOS 20.3.2 with local changes.

[image]

I don't expect this to be enough info to debug - what can I do to see why the detection fails, so I can make a minimum viable sample to reproduce?

vrepetenko, 2024-05-23:
Hi, 
Supported ChibiOS RT version is 7.x or higher, so you need to have ChibiOS v21.x.x

Woodgnome, 2024-05-26:
> Here is the guide for getting started TODO: add link to the guide.md

Has a guide been written since the last update to the README or would I have to reverse engineer it from the current ChibiOS file if I wanted to try adding support for an older version?

PhilippHaefele, 2024-05-26:
> > Here is the guide for getting started TODO: add link to the guide.md
> 
> 
> 
> Has a guide been written since the last update to the README or would I have to reverse engineer it from the current ChibiOS file if I wanted to try adding support for an older version?

The guide is to be written. So for now some "reverse engineering" is needed.

Two things to keep in mind:
- Should we have one codebase for both versions or two separate ones
- How are the different versions detected reliable

Woodgnome, 2024-05-31:
> The guide is to be written. So for now some "reverse engineering" is needed.

Fair enough. Currently no specific plans to look at it, so it may be weeks/months when (or if) I get tired of debugging without the OS awareness.