mcu-debug/rtos-views #55: Can't see the run-time in freertos ?

woquchonglang, 2024-11-30:
I have add a tim to add the FreeRTOSRunTimeTicks, but I can't see the runtime yet?

[image]

[image]

haneefdm, 2024-11-30:
Have you done everything that is needed?

https://www.freertos.org/Documentation/02-Kernel/02-Kernel-features/08-Run-time-statistics

Also, did you check in your FW or debugger that you are actually collecting the data?

malsyned, 2025-02-08:
I'm having this same issue, and I believe I've confirmed that everything should be working.

I have an example task that is periodically printing the results of `vTaskGetRunTimeStatistics()` to a serial port, and the output looks correct to me:
```
example         2173981         7%
IDLE            28295547                92%
```

cortex-debug is showing the relevant timing data is being updated in memory:

[image]

But my Runtimes are still just showing up as `??.??%`.

[image]

I wonder, could this could be related to `ulTotalRunTime` changing from a scalar to an array in FreeRTOS V11.0.0 to support SMP? That was released a little over a year ago and has this change:

```diff
commit ae3a498e435cecdb25b889f2740ea99027dd0cb1
diff --git a/tasks.c b/tasks.c
-PRIVILEGED_DATA static volatile configRUN_TIME_COUNTER_TYPE ulTotalRunTime = 0UL; /**< Holds the total amount of execution time as defined by the run time counter clock. */
+PRIVILEGED_DATA static volatile configRUN_TIME_COUNTER_TYPE ulTotalRunTime[ configNUMBER_OF_CORES ] = { 0U }; /**< Holds the total amount of execution time as defined by the run time counter clock. */

```

FreeRTOS Version: V11.1.0
Architecture: ARM Cortex-R4F
MCU: TI RM44L520PGE
Compiler: clang version 18.1.3 (LLVM-embedded-toolchain-for-Arm)

malsyned, 2025-02-10:
@woquchonglang I don't know if your problem has the same root cause as mine, but I opened PR #57 which fixes my problem.

woquchonglang, 2025-02-12:
@malsyned  it works!