mcu-debug/rtos-views #61: FreeRTOS: hints always report missing info, even when info is found and displayed

malsyned, 2025-04-14:
In testing out the new FreeRTOS queue support from @hwmland's PR #45, I found that even after successfully configuring the queue registry and adding queues to the registry, I still got a hint reading 
> Missing Queue/Mutex/Semaphore info..:
> Register queues/semaphores/mutexes of interest using vQueueAddToRegistry()

[image]

This looks to me like it's because `RTOSFreeRTOS.createHmlHelp()` is idempotent: it only changes `this.helpHtml` the first time it's called, which is in the first call to `getThreadInfo()` from `refresh()`. That happens before `getQueueInfo()` is ever called, so `this.queueInfo` hasn't been populated and will always be `[]`.

This isn't completely trivial to fix, because `createHtmlHelp()` uses an `RTOSStrToValueMap` that's local to `getThreadInfo()`. I think it is fixable, but it'll require a more extensive reworking of how the help text is created and when it is updated.

haneefdm, 2025-04-14:
@hwmland Do you have some time to take a look?

hwmland, 2025-04-15:
Sorry, not now. Due to workload I still didn't manage even to reinstall RpiPico toolchain yet after my disk crash :( 
Isn't it enough to comment one line (see comment in this code fragment):
```ts
    protected createHmlHelp(th: RTOSCommon.RTOSThreadInfo, thInfo: RTOSCommon.RTOSStrToValueMap) {
        function strong(s: string) {
            return `<strong>${s}</strong>`;
        }
        if (this.helpHtml === undefined) { // isn't it enough to comment this line out?
            this.helpHtml = '';
            try {
                let ret = '';
                if (!thInfo['uxTCBNumber'].val) {
// code continues here...
```

haneefdm, 2025-04-15:
It is also not super important for the help to be super context sensitive. It depends on the wording of the Help text.