mcu-debug/rtos-views #62: Missing "Hints to get more..." link

bmcdonnell-fb, 2025-05-06:
### Annotated example from readme

[image]

### Mine, missing link at the bottom:

[image]

I know I need to `#define configCHECK_FOR_STACK_OVERFLOW 2` (and maybe also `#define configRECORD_STACK_HIGH_ADDRESS 1`) to show values instead of `???`s. But why is the hints link missing?

### My setup

VS Code: 1.99.3
Windows 10, 22H2 (OS Build 19045.5737)
Cortex-Debug: 1.12.1
mcu-debug RTOS Views: 0.0.7
Probe: J-Trace
J-Link: 8.10e
FreeRTOS: 10.3.1
Debugger: arm-none-eabi toolchain v14.2.rel1

haneefdm, 2025-05-06:
If you click on the area where the link is supposed anything happens? Or is it just dead space.

Just for kicks, can you change your VSCode theme temporarily to see if the link is visible? Also, change the size of the window (up or down a few pixels) to see if it shows up.

We tried to follow the current theme everywhere but...

bmcdonnell-fb, 2025-05-07:
Weird, I'm seeing it now, when `configRECORD_STACK_HIGH_ADDRESS` is not defined. I guess it only shows when that is the case?

I think I've resized the panel since I submitted this issue.

Is the following statement I made wrong? It seems I only need to set `configRECORD_STACK_HIGH_ADDRESS` to 1.

> I know I need to `#define configCHECK_FOR_STACK_OVERFLOW 2` ... to show values instead of `???`s.

haneefdm, 2025-05-07:
The hints only shown in an (feeble) attempt to try to explain missing information for certain columns. There are too many combinations of config's possible, and a lot of users don't know what any of them do or how it impacts memory/perf. Also, they start with bare minimum enable..

> Is the following statement I made wrong? It seems I only need to set configRECORD_STACK_HIGH_ADDRESS to 1.

I don't remember anymore...the exact dependencies. Haven't done any FreeRTOS work lately.

bmcdonnell-fb, 2025-05-07:
Ah, I found [the relevant function](https://github.com/mcu-debug/rtos-views/blob/9a540c50581eb9209d198d5d796e52cf946980c8/src/rtos/rtos-freertos.ts#L239-L294). It does report [if `configRECORD_STACK_HIGH_ADDRESS` isn't defined](https://github.com/mcu-debug/rtos-views/blob/9a540c50581eb9209d198d5d796e52cf946980c8/src/rtos/rtos-freertos.ts#L252). `configCHECK_FOR_STACK_OVERFLOW` isn't mentioned anywhere, so apparently I was wrong about that.

Closing since I can't reproduce the issue. Thanks.

closed 2025-05-07 by bmcdonnell-fb