mcu-debug/rtos-views #65: FreeRTOS: "Stack Used" inaccurate when configENABLE_MPU is 1 on ARM

malsyned, 2025-08-02:
On many ARM ports, when the `configENABLE_MPU` is used, `pxTopOfStack` doesn't actually point to the top of the stack. It points into a separate `pTCB->xMPUSettings.ulContext[]` array that is used to store the task context for context switches. As a result, computing **Stack Used** as the difference between `pxTopOfStack` and either `pxStack` or `pxEndOfStack` gives incorrect, meaningless results.

[image]

I discovered this on the ARM_CRx_MPU port, but a quick grep show that at least the following ports are likely affected:

* ARMv8M
* ARM_CM0
* ARM_CM23
* ARM_CM33
* RP2350_ARM
* ARM_CM35P
* ARM_CM3_MPU
* ARM_CM4_MPU
* ARM_CM55
* ARM_CM85
* ARM_CRx_MPU
* ARM_CM4F_MPU

I believe in all cases the actual top-of-stack value is stored in ulContext (with a caveat for non-privileged tasks on the ARM_CRx port), but it may not be in the same place on every port. It looks like there's two different ulContext layouts employed.

Layout 1: ARMv8M, ARM_CM23, ARM_CM33, ARM_CM35P, ARM_CM55, ARM_CM85, RP2350_ARM

Layout 2: ARM_CM0, ARM_CM3_MPU, ARM_CM4_MPU, ARM_CM4F_MPU

Layout 3: ARM_CRx_MPU. The extra caveat on this port is that the task stack pointer is in ulContext for privileged tasks and for non-privileged tasks that were preempted, and in `pTCB->xMPUSettings.xSystemCallStackInfo.pulSystemCallStackPointer` for non-privileged tasks that used a system call to context-switch.

I'm not really sure what the right way is to approach this issue. It might be possible to introspect into which fields are defined in various structs to figure out whether a separate MPU context is being used and, if so, where to find the task's stack pointer. I think unfortunately in most cases the actual knowledge of where a task's stack pointer is saved is only represented implicitly in `pxPortInitialiseStack()`, `portSAVE_CONTEXT`, and `portRESTORE_CONTEXT`, so a fix for this would probably have to settle for heuristics based on existing practice.

A more blunt partial fix that would at least prevent the display of misleading information could be to disable the **Stack Used** column if a symbol introspection indicated that an MPU context array was being used.

(Note that this is all talking about suspended tasks. Running tasks probably have stale top-of-stack values. See also #28)

malsyned, 2025-08-06:
I did some more digging into this, and it looks like it is possible to get the task's stack pointer from the ulContext array, but it's messy because the location depends on various compile-time constants.

### Layout 1

What I called "Layout 1" is actually this collection of possible layouts with different blocks either present or absent, documented in the `portmacrocommon.h` of all of the ports in question:

```c
    #if ( ( configENABLE_FPU == 1 ) || ( configENABLE_MVE == 1 ) )

        #if ( ( configENABLE_TRUSTZONE == 1 ) && ( configENABLE_PAC == 1 ) )

            /*
             * +-----------+---------------+----------+-----------------+------------------------------+------------+-----+
             * |  s16-s31  | s0-s15, FPSCR |  r4-r11  | r0-r3, r12, LR, | xSecureContext, PSP, PSPLIM, | TaskPacKey |     |
             * |           |               |          | PC, xPSR        | CONTROL, EXC_RETURN          |            |     |
             * +-----------+---------------+----------+-----------------+------------------------------+------------+-----+
             *
             * <-----------><--------------><---------><----------------><-----------------------------><-----------><---->
             *      16             17            8               8                     5                     16         1
             */
            #define MAX_CONTEXT_SIZE    71

        #elif ( ( configENABLE_TRUSTZONE == 1 ) && ( configENABLE_PAC == 0 ) )

            /*
             * +-----------+---------------+----------+-----------------+------------------------------+-----+
             * |  s16-s31  | s0-s15, FPSCR |  r4-r11  | r0-r3, r12, LR, | xSecureContext, PSP, PSPLIM, |     |
             * |           |               |          | PC, xPSR        | CONTROL, EXC_RETURN          |     |
             * +-----------+---------------+----------+-----------------+------------------------------+-----+
             *
             * <-----------><--------------><---------><----------------><-----------------------------><---->
             *      16             17            8               8                     5                   1
             */
            #define MAX_CONTEXT_SIZE    55

        #elif ( ( configENABLE_TRUSTZONE == 0 ) && ( configENABLE_PAC == 1 ) )

            /*
             * +-----------+---------------+----------+-----------------+----------------------+------------+-----+
             * |  s16-s31  | s0-s15, FPSCR |  r4-r11  | r0-r3, r12, LR, | PSP, PSPLIM, CONTROL | TaskPacKey |     |
             * |           |               |          | PC, xPSR        | EXC_RETURN           |            |     |
             * +-----------+---------------+----------+-----------------+----------------------+------------+-----+
             *
             * <-----------><--------------><---------><----------------><---------------------><-----------><---->
             *      16             17            8               8                  4                16         1
             */
            #define MAX_CONTEXT_SIZE    70

        #else /* if ( ( configENABLE_TRUSTZONE == 1 ) && ( configENABLE_PAC == 1 ) ) */

            /*
             * +-----------+---------------+----------+-----------------+----------------------+-----+
             * |  s16-s31  | s0-s15, FPSCR |  r4-r11  | r0-r3, r12, LR, | PSP, PSPLIM, CONTROL |     |
             * |           |               |          | PC, xPSR        | EXC_RETURN           |     |
             * +-----------+---------------+----------+-----------------+----------------------+-----+
             *
             * <-----------><--------------><---------><----------------><---------------------><---->
             *      16             17            8               8                  4              1
             */
            #define MAX_CONTEXT_SIZE    54

        #endif /* #if ( ( configENABLE_TRUSTZONE == 1 ) && ( configENABLE_PAC == 1 ) ) */

    #else /* #if ( ( configENABLE_FPU == 1 ) || ( configENABLE_MVE == 1 ) ) */

        #if ( ( configENABLE_TRUSTZONE == 1 ) && ( configENABLE_PAC == 1 ) )

            /*
             * +----------+-----------------+------------------------------+------------+-----+
             * |  r4-r11  | r0-r3, r12, LR, | xSecureContext, PSP, PSPLIM, | TaskPacKey |     |
[... 116 lines cut ...]
     * | ulCriticalNesting | FPSCR |  S0-S31  | R0-R12 | SP (R13) | LR (R14) | PC (R15) | CPSR |
     * +-------------------+-------+----------+--------+----------+----------+----------+------+
     *
     * <------------------><------><---------><--------><---------><--------><----------><----->
     *           1            1        32         13         1         1          1        1
     */
    #define CONTEXT_SIZE 51U
#else
    /*
     * +-------------------+--------+----------+----------+----------+------+
     * | ulCriticalNesting | R0-R12 | SP (R13) | LR (R14) | PC (R15) | CPSR |
     * +-------------------+--------+----------+----------+----------+------+
     *
     * <------------------><--------><---------><--------><----------><----->
     *           1             13         1         1          1        1
     */
    #define CONTEXT_SIZE 18U
#endif /* CONTEXT_SIZE */
```

### Analysis

The size of `ulContext` alone can be used to infer the position of the stack pointer in all but two of these layouts.

Unfortunately, if `ulContext` holds 20 ints, it could be a CM0 or a CM3, which use different stack layouts. There are other clues as to which device is being run, though, including whether `xMPUSettings` contains an `xRegion` member.

haneefdm, 2025-08-07:
Thanks for all the work. Did not know that using an MPU can get this involved :-)

Just FYI: so far, we have not made assumptions that the RTOS is running on an ARM MCU. It could be something else like RISC-V or Xtensa, etc. Of course, this MPU is probably specific for CMx/CRx devices.

While we have all of these layouts, what does the ABI say? Can that help? GDB will always use the ABI to interpret the stack, so could we use GDB's knowledge to our advantage? It may make sense that any RTOS would also use the ABI to save/restore contexts

Note that there is no assumption that gdb is the debugger. We rely on the debugger supporting the [Debug Adapter Protocol](https://microsoft.github.io/debug-adapter-protocol//) defined by Microsoft

malsyned, 2025-08-07:
> Just FYI: so far, we have not made assumptions that the RTOS is running on an ARM MCU. It could be something else like RISC-V or Xtensa, etc. Of course, this MPU is probably specific for CMx/CRx devices.

The FreeRTOS does have a few non-ARM ports that seem to have MPU support, or at least test for `configENABLE_MPU`: TriCore_1782, TriCore_38xa, Xtensa_ESP32.

However, they don't appear to use separate context storage in `xMPUSettings.ulContext[]`. It looks to me like the practice of storing task context in the task control block and pointing `pxTopOfStack` into it instead of into the task's stack is unique to ARM ports with `configENABLE_MPU == 1`.

This is a difficult situation to future-proof for. FreeRTOS ports can be a little wild-west sometimes. The safest bet seems to me that if ever another port did use separate context storage, it would be more likely to use the `xMPUSettings.ulContext[]` convention than anything else, although what the layout of that array would be is anybody's guess. 🤷 

> While we have all of these layouts, what does the ABI say? Can that help? GDB will always use the ABI to interpret the stack, so could we use GDB's knowledge to our advantage? It may make sense that any RTOS would also use the ABI to save/restore contexts

I dug around in the AARCH32 ABI documentation, but I didn't see anything that addressed this directly. The ABI specifies which registers are caller-saved vs. callee-saved, but it's silent on what order they get saved in. Most ports do seem to be treating the "Argument registers" `r0-r3` as a group and "Variable registers" `r4-r11` as a group, but beyond that, there is plenty of variation. On non-MPU Cortex-Rx ports, I found no fewer than 3 different incompatible stack layouts, so I'm not super optimistic that there's a standard here -- at least one that is being consistently followed.

Is it possible to ascertain via DAP the architecture (e.g. "arm") or even better, the exact ARM version (e.g. "armv7m", "armv8m")? If it is, then this becomes potentially easier. The breakdown of stack layouts is:

### ARMv6M, ARMv8M

```
#if configENABLE_FPU
    s16-s31
    s0-s15
    FPSCR
#endif
r4-r11
r0-r3
r12
LR
PC
xPSR
#if configENABLE_TRUSTZONE
    xSecureContext (can check for xSecureContext global variable)
#endif
PSP
#if ARMv8M
    PSPLIM
#endif
CONTROL
EXC_RETURN
#if configENABLE_PAC
    TaskPacKey (can check for presence of vApplicationGenerateTaskRandomPacKey function)
#endif
(dummy word)
```

### ARMv7M

```
#if configENABLE_FPU
    s16-s31
    s0-s15
    FPSCR
#endif
CONTROL
r4-r11
EXC_RETURN
PSP
r0-r3
r12
LR
PC
xPSR
(dummy word)
```

### ARMv7R

```
ulCriticalNesting
#if configENABLE_FPU
    FPSCR
    s0-s31
#endif
R0-R12
SP
LR
PC
CPSR
(dummy word)
```

***

The optional block of floating point registers at the beginning of most of these context blocks is a problem because I don't know of any way at runtime to tell whether `configENABLE_FPU == 1`. It has a per-port default, but many ports let you override it to disable saving FPU context on MCUs with FPUs. However, we can figure out how far from the *end* of `ulContext` the stack pointer is if we know the ARM version by doing the following:

* ARMv6M? `ulContext.length - 4`
* ARMv7M? `ulContext.length - 10`
* ARMv7R? `ulContext.length - 5`
* ARMv8M?
    ```
    let spIndex = ulContext.length - 4
    global function symbol vApplicationGenerateTaskRandomPacKey exists?
        spIndex += 16 // Except I think the comments are actually wrong and this should be += 4
    ````

This logic would all only run if `xMPUSettings.ulContext[]` was true. If there's ever an architecture that had this struct field but wasn't ARM of a known ARM version, we could fall back to ??? for Stack Used/Free so that we wouldn't be providing incorrect information.

**But**, of course, all of this only works if we have a way via DAP to access the target architecture and arch version, and I don't know if that's possible. Do you?

***

FYI I have a PR pretty much done that performs ARM architecture inference based just on stack sizes. It looks like this:

```typescript
        let spIndex: number;
        if (ulContext.length === 20) {
            // Layout not unambiguously determined by context size, fall
            // back to heuristic based on port struct differences.
            if ('xRegion' in xMPUSettings) {
                // ARMv7-M, No FPU (Cortex-M3)
                spIndex = 10;
            } else {
                // ARMv6-M (Cortex-M0)
                spIndex = 8 + 8;
            }
        } else {
            const spContextIndices: { [key: number]: number } = {
                //// ARM-v8M ////
                // Layouts described in:
                //   ARMv8M/non_secure/portmacrocommon.h
                //   {GCC,IAR}/
                //     {ARM_CM23,ARM_CM33,ARM_CM35P,ARM_CM55,ARM_CM85}/
                //     {ARM_CM23,ARM_CM33,ARM_CM35P,ARM_CM55,ARM_CM85}_NTZ/
                //       non_secure/portmacrocommon.h
                // FPU, TrustZone
                71: 16 + 17 + 8 + 8 + 1,
                55: 16 + 17 + 8 + 8 + 1,
                // FPU, No TrustZone
                70: 16 + 17 + 8 + 8,
                54: 16 + 17 + 8 + 8,

                // No FPU, TrustZone
                38: 8 + 8 + 1,
                22: 8 + 8 + 1,
                // No FPU, No TrustZone
                37: 8 + 8,
                21: 8 + 8,

                //// ARM-v7EM ////
                // Layout described in {GCC,IAR,RVDS}/ARM_CM4F?_MPU/portmacro.h
                53: 16 + 17 + 10,

                //// ARM-v7R ////
                // Layouts described in GCC/ARM_CRx_MPU/portmacro_asm.h
                // FPU
                51: 1 + 1 + 32 + 13,
                // No FPU
                18: 1 + 13,
            };

            spIndex = spContextIndices[ulContext.length];
        }
```

haneefdm, 2025-08-07:
> But, of course, all of this only works if we have a way via DAP to access the target architecture and arch version, and I don't know if that's possible. Do you?

I am not near my computer/boards to try/test anything, but could you detect it by reading the CPUID register or the MIDR (Main ID Register)? I am not even sure that the CPUID is at a fixed location, though.

malsyned, 2025-08-07:
It looks like CPUID is at `0xE000ED00` on every Cortex-M CPU with an MPU. I checked the reference docs for all of the applicable Cortex-M types.

It doesn't exist on Cortex-R, though. On Cortex-R, there is an MIDR coprocessor register, but the only way I know how to read that is by sending `monitor cp15 0xc0, 0xc0, 0, 0` to JLinkGDBServer via GDB. Unless you know a way to do this with DAP that is agnostic to the debugger or debug server, it may be better just to use a read failure on `0xE000ED00` identify Cortex-R?

Alternatively, we could use the presence or absence of certain members of `xMPUSettings` to detect which context layout to use.

|                 | ARM-v6M | ARM-v7M | ARM-v7R | ARM-v8M |
|-----------------|---------|---------|---------|---------|
| ulMAIR0         |         |         |         | ✅       |
| xRegion         |         | ✅       | ✅       |         |
| xRegionSettings | ✅       | ✅       |         | ✅       |

haneefdm, 2025-08-07:
Yes, CPUID is for the M-profile. The A and R profiles used MIDR AFAIR

So you can't just evaluate the expression  `*((unsigned int *)0xE000ED00)`, handle failures, then use alternate addresses for MIDR, etc.?

haneefdm, 2025-08-07:
But you can read the CPUID using the debuggers evaluate expression. Even if we can address the M class MCUs that is a big deal.

We cannot rely on any GDB server. We don't have access to gdb servers, and no, I don't want to use 'monitor' commands either ... because that implies `gdb`. This extension also works with the ARM debugger, which does not have gdb or a gdb server in a traditional sense. Some use Microsoft's debugger

haneefdm, 2025-08-07:
Btw, the design goal of this extension is to work with various debuggers. I also maintain cortex-debug, and I split this out of that extension to create a standalone version for that purpose.

malsyned, 2025-08-07:
> But you can read the CPUID using the debuggers evaluate expression. Even if we can just address the M class MCUs that is a big deal.

I agree. If you're good with assuming that "couldn't read 0xE000ED00" means "We're on Cortex-R", then I think that's a completely workable solution. Does that sound good to you?

malsyned, 2025-08-07:
> Btw, the design goal of this extension is to work with various debuggers. I also maintain cortex-debug, and I split this out of that extension to create a standalone version for that purpose.

Yup. absolutely, I'm on-board with that. I'm just flagging areas that I don't know how to solve cleanly in a debugger-agnostic way, in case you know a trick that I don't.

Is `*((unsigned int *)0xE000ED00)` guaranteed to work with any DAP debugger? If so, and if you are OK with assuming "couldn't read 0xE000ED00" means "We're on Cortex-R", then I don't see any problems with going that direction with this.

haneefdm, 2025-08-08:
>  I don't know how likely ARM is to create new 32-bit ARM cores anyway

Unlikely, but you never know. I thought they were done with the M55, but then came M52 and M85. Also variants like M0+, M33P pop up

> If so, and if you are OK with assuming "couldn't read 0xE000ED00" means "We're on Cortex-R", then I don't see any problems with going that direction with this

I think we should determine if this is an ARM processor first. Say, if the link register (LR) exists. Or some other way.

```
if this is an ARM
    if CPUID returns something meaningful then
        it is ARM M profile, and CPUID can tell you exactly which one ONLY if needed
    else
        assume R profile
```

malsyned, 2025-08-11:
> I think we should determine if this is an ARM processor first. Say, if the link register (LR) exists. Or some other way.

I've been treating the presence of `xMPUSettings.ulContext` as the test for whether we're on ARM with an MPU. But I suppose just because only ARM ports use `ulContext` today doesn't mean other ports couldn't in the future. The link @aggarg posted about the kernel threat model isn't ARM-specific, so when it says

> Task context is stored in the Task Control Block (TCB), which is a kernel internal data structure, and so not accessible to unprivileged code

Any future non-ARM ports that adhere to that mitigation guidance could suddenly sprout a `ulContext` array.

I like your register-presence heuristic approach. Instead of `$lr`, could we use `$xpsr` to detect Cortex-M and `$cpsr` to detect Cortex-R? That would avoid the need to attempt a read of CPUID on non-M targets. So it would be:

```
If has xMPUSettings.ulContext:
    If has $cpsr:
        Assume Cortex-R profile (Not Cortex-A because we wouldn't get this far)
    else if has $xpsr:
        Assume Cortex-M profile
        Read CPUID from 0xE000ED00
    else:
        Assume non-ARM (unsupported for now)
else:
    `pxTopOfStack` is stack pointer
````

For Cortex-M we could use `$primask`, `$basepri`, or `$faultmask` instead of `$xpsr`, I just picked `$xpsr` because it rhymes with `$cpsr`. 🤷

malsyned, 2025-08-11:
@haneefdm how does something like this look to you?

```typescript
        let spIndex: number;
        const registers = await this.getRegisters(frameId) || {};
        if ('xpsr' in registers) {
            // ARM Cortex-M
            const cpuid = parseInt(await this.getExprVal('*(unsigned int *)0xe000ed00', frameId) || '');
            const implementer = cpuid >> 24 & 0xff;
            const arch = cpuid >> 16 & 0xf;
            const partNo = cpuid >> 4 & 0xfff;
            if (implementer !== 0x41) {
                return null;
            }
            if (arch === 0xf && (partNo & 0xf00) === 0xc00) {
                // Cortex-M3 or Cortex-M4 (ARM-v7M)
                spIndex = 10;
            } else {
                // ARM-v6M, ARMv8M
                spIndex = 16;
                if (this.xSecureContext) {
                    spIndex++;
                }
            }
        } else if ('cpsr' in registers) {
            // ARM Cortex-R
            spIndex = 14;
        } else {
            return null;
        }

        if (ulContext.length >= 50) {
            // configENABLE_FPU == 1
            spIndex += 33;
        }
        return parseInt(ulContext[spIndex].val);
```

malsyned, 2025-08-11:
> const registers = await this.getRegisters(frameId) || {};
> 
> How are you doing `getRegisters()`? Not part of the protocol, I don't think. If you want a particular register, you can use the expression `$xpsr ` (maybe uppercase). This will work with gdb but not sure about others ... and then we go generic fall back.

I'm getting the list of variable scopes using the DAP `scopes` request, then looking for a scope whose down-cased name contains the string "registers". I did it that way to avoid the GDB-specific dependency on `$regname` syntax.

```typescript
    // in rtos-freertos.ts
    private async getRegisters(frameId: number): Promise<RTOSCommon.RTOSStrToValueMap | null> {
        let scopes: DebugProtocol.Scope[];
        try {
            scopes = await this.getScopes(frameId);
        } catch (e) {
            return null;
        }
        for (const scope of scopes) {
            if (scope.name.toLowerCase().match('registers')) {
                return this.getVarChildrenObj(scope.variablesReference, 'Registers');
            }
        }
        return null;
    }

    // in rtos-common.ts
    protected async getScopes(useFrameId: number): Promise<DebugProtocol.Scope[]> {
        const arg: DebugProtocol.ScopesArguments = {
            frameId: useFrameId
        };

        const result = await this.customRequest('scopes', arg);
        if (!result || !result.scopes || !result.scopes.length) {
            throw new Error('Failed to retrieve variable scopes');
        }

        return result.scopes;
    }
```

haneefdm, 2025-08-11:
FYI: Not everyone uses scopes for Registers. This was my invention/hack because I couldn't convince MS to get a notification when the Stack Window changed state. Even Global/Statics was our invention I know for a fact other don't do this as it is very tricy/expensive

Maybe others copied what I did in cortex-debug but I can't be sure. They may have named that scope differently as well.

I need to think about it. The only debugger I am very interested in right now is the ARM debugger but I don't have a license/copy.

haneefdm, 2025-08-12:
Let's do it the way you have it with some tweaks. Please note that we should not be calling `getRegisters()` for every refresh. Only during the discovery phase. It appears you are calling it for every refresh. Cache what you need to. Also, do not rely on the values of the registers. Just names for the most part. Note that there can be format differences beyond hex vs decimal. For instance, at one time, Cortex-Debug was showing bit fields of the status-like registers.

Please add a comment for getRegisters() that it is not guaranteed to work since we are relying on current convention rather than the DAP Spec.

Now, about your discovery code. See my comments below if your method needs some refinement.
```JavaScript
        if ('xpsr' in registers) {
            // ARM Cortex-M
            // haneefdm: Is this true? Cortex Cortex-R5, R8, R52 also seem to have `xpsr` (source Google)
            // haneefdm: However, if xpsr exists and cpuid exists then we can be convinced that it is Cortex-M?!?!
            // Also, can getting cpuid fail for whatever reason? We should guard against that.
             ...
        } else if ('cpsr' in registers) {
            // ARM Cortex-R
            spIndex = 14;
        } else {
            return null;
        }

        if (ulContext.length >= 50) {
            // configENABLE_FPU == 1
            spIndex += 33;
        }
```

haneefdm, 2025-08-12:
Thanks btw, for all this work.

malsyned, 2025-08-12:
> Let's do it the way you have it with some tweaks.

OK great. I'll put together a PR for you on this with your tweak requests addressed. I hope to have it to you later today.

>             // haneefdm: Is this true? Cortex Cortex-R5, R8, R52 also seem to have `xpsr` (source Google)
>             // haneefdm: However, if xpsr exists and cpuid exists then we can be convinced that it is Cortex-M?!?!

Can you send me on any information you have about xpsr on Cortex-R? I haven't been able to find any evidence of an xpsr register on R5, R52, or R8. I dug into the TRMs for those architectures, and I find that they have a CPSR, banked SPSRs, R8 has an APSR which is a different view into the CPSR. But I don't see a register called "xpsr" like on the M profile in any of them. Is it possible that you were seeing documentation referring to the collection of CPSR/SPSR/APSR as "xPSR"? I don't think it's an official register name on those profiles.

I'm using a few other sources of information as well to make inferences.

#### SEGGER JLinkGDBServer target description XML

Their document "UM08030_JLinkGDBThreadAwareness.pdf" contains copies of the target description files that JLinkGDBServer transmits to GDB, and they only have xpsr in the M-profile descriptions. They have a single description file titled "Cortex-A, Cortex-R" that lists only a "cpsr" and the banked versions of "spsr".

#### GDB target description XML in gdb/features/arm

In GDB's source tree, it defines cpsr as a register 25 in arm-core.xml
```xml
  <reg name="cpsr" bitsize="32" regnum="25"/>
```
and then in arm-m-profile.xml and arm-m-profile-with-fpa.xml it defines xpsr as register 25 
```xml
<reg name="xpsr" bitsize="32" regnum="25"/>
```

#### FreeRTOS ARM_CRx_MPU port

This port is for all R-profile MCUs with an MPU including R5, R52, and R8. It stores CPSR as part of its context (actually the SPSR copy saved when the SVC exception fired), but it doesn't have anything called XPSR saved to its context. This is in contrast to all of the M-profile ports, which save XPSR, PRIMASK, BASEPRI, and FAULTMASK to the task context block instead.

`git grep --recurse-submodules  -i xpsr FreeRTOS/Source/portable/GCC/ARM_CRx_MPU/` doesn't turn up any results.

>             // Also, can getting cpuid fail for whatever reason? We should guard against that.

I'm already guarding against it containing an implausible value by matching the high byte against the ARM manufacturer code. (in my latest version, it's matching against four different possible manufacturer codes instead of just 0x41).

I suppose it could also fail in a way that generates an exception. I can make sure to guard against that in my PR.

> Thanks btw, for all this work.

You're welcome!

malsyned, 2025-08-12:
> Doesn't mean Google is right. Turns out while those three devices may have an xPSR register, it is not a "core register". Google "do the cortex-r5 r8 r52 have a core register xpsr"

I'm doing that search, but it's not turning up anything useful for me. I have AI results turned off, though, so if that's the only place you're seeing the assertion that there's a Cortex-R with a register named xpsr, maybe that could be why?

haneefdm, 2025-08-13:
You probably figured out most of the following:

`tryDetect()` is for detecting if an RTOS is present. It can do whatever it wants to determine if that particular RTOS exists.

`getVarIfEmpty()` is a helper for evaluating variables - expressions actually. It is used by tryDetect while determining that an OS exists.

One thing to remember is that things can fail rather abruptly while we are getting values of expressions, expanding children, etc. This occurs during initial discovery (tryDetect) and during refresh. Many times, the debugger pauses just for a millisecond or less (set a breakpoint while running, initial connect/attach/reset, break-on-variable) and continues and we don't get enough time to do all our queries. Also true if the user is single-setting furiously. Nothing much works if the FW is running, and it can change abruptly and asynchronously. We have no control.

So, for scopes, you have to do something similar and allow/prepare for retries. For instance
1. Only call customRequest() if FW status is in the stopped state. Note that some requests will succeed even if FW is running. Scopes, stack, variable, expression queries will all fail.
2. If it fails, we have to determine if it is because the FW got busy (running) right after we called for scopes. If so, try again. Scopes should never fail if FW is stopped. It can be empty, though.

This is a flaw in the DAP original protocol API calls. I fought for it to be fixed, and it did get fixed but too late. Can't rely on it because too many debuggers already existed, and the fix was optional.

malsyned, 2025-08-13:
> One thing to remember is that things can fail rather abruptly while we are getting values of expressions, expanding children, etc. This occurs during initial discovery (tryDetect) and during [...]

Thanks, this explanation helps a lot, actually. So I think that actually, I shouldn't be doing the register name collecting or architecture detection in `tryDetect()`, since they have nothing to do with probing to detect the RTOS. Instead I should be doing some caching and retry logic in `refresh()`, do you agree?

haneefdm, 2025-08-14:
Oh no, I missed a couple of updates here. `tryDetect` is for trying to detect the OS. It will be called repeatedly until either the OS is known not to exist or more work needs to be done to detect the OS. `refresh()` is called once we KNOW that the OS exists.

So, your detection of the architecture should be in the `tryDetect`

haneefdm, 2025-08-14:
> Since tryDetect() stops being called once an RTOS is detected, isn't it possible for an optional global like ulTotalRunTime and xQueueRegistry to fail to update due to transient reasons, then never be checked again because it was optional and so tryDetect() succeeded?

Oh yeah. Anything can fail during a `refresh()` so we abort the entire `refresh()` and try again the next time the FW stops. The user can also manually trigger a `refresh()`. In reality, `refresh()` does fail many times during fast single stepping but we don't panic. The user eventually takes a breather with the FW stopped and we do a full refresh(). This is why we print the date/time we did a full `refresh()`. Partial refreshes are never reported...too much noise.

If you noticed, a `refresh()` can take upto a second or more. A lot of stuff can happen in a second.

Btw, do not try to refresh RTOS state/variables (doing debugger queries) during a `getHTML()`. That is not the intent. `getHTML()` should just render and optionally cache the HTML(). We also try to warn that the refreshed info might be incomplete

https://github.com/mcu-debug/rtos-views/blob/09f25fdd7837e31feb3e0a065ffc0abbb5d307ae/src/rtos/rtos-freertos.ts#L830

Its been a long time since I wrote this code and others have updated since. Please double check what I am saying.

closed 2025-08-30 by haneefdm