mcu-debug/rtos-views #71: XRTOS tab - task loading on FreeRtos leads to infinite loop

156898293, 2025-09-01:
I just got version 0.0.10 auto installed on my system and it fails to load tasks properly. When I revert back to 0.0.7 it seems to work again.
The problem seems to be in infinite loop via gdb:
--> R (variables-389): {"type":"response","request_seq":389,"success":true,"command":"variables","body":{"variables":[{"name":"xItemValue","value":"","type":"TickType_t","evaluateName":"(((((((((((((((((((((((((struct xLIST_ITEM *)0x92900)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->xItemValue","variablesReference":0},{"name":"pxNext","value":"","type":"struct xLIST_ITEM *","evaluateName":"(((((((((((((((((((((((((struct xLIST_ITEM *)0x92900)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxNext","variablesReference":1899},{"name":"pxPrevious","value":"","type":"struct xLIST_ITEM *","evaluateName":"(((((((((((((((((((((((((struct xLIST_ITEM *)0x92900)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious","variablesReference":1900},{"name":"pvOwner","value":"","type":"void *","evaluateName":"(((((((((((((((((((((((((struct xLIST_ITEM *)0x92900)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pvOwner","variablesReference":0},{"name":"pvContainer","value":"","type":"struct xLIST *","evaluateName":"(((((((((((((((((((((((((struct xLIST_ITEM *)0x92900)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pxPrevious)->pvContainer","variablesReference":1901}]},"seq":5053}
<--   C (variables-390): {"command":"variables","arguments":{"variablesReference":1882},"type":"request","seq":390}
--> E (output): {"type":"event","event":"output","body":{"category":"console","output":"1: (13830) <-1414-var-list-children --simple-values \"var92.5.xHandle.xTasksWaitingToSend.xListEnd.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious\" 0 1000\n"},"seq":5056}
1: (13830) <-1414-var-list-children --simple-values "var92.5.xHandle.xTasksWaitingToSend.xListEnd.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious" 0 1000
--> E (output): {"type":"event","event":"output","body":{"category":"console","output":"1: (13831) ->1414^done,numchild=\"5\",children=[child={name=\"var92.5.xHandle.xTasksWaitingToSend.xListEnd.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.xItemValue\",exp=\"xItemValue\",numchild=\"0\",value=\"\",type=\"TickType_t\"},child={name=\"var92.5.xHandle.xTasksWaitingToSend.xListEnd.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxNext\",exp=\"pxNext\",numchild=\"5\",value=\"\",type=\"struct xLIST_ITEM *\"},child={name=\"var92.5.xHandle.xTasksWaitingToSend.xListEnd.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious\",exp=\"pxPrevious\",numchild=\"5\",value=\"\",type=\"struct xLIST_ITEM *\"},child={name=\"var92.5.xHandle.xTasksWaitingToSend.xListEnd.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pvOwner\",exp=\"pvOwner\",numchild=\"0\",value=\"\",type=\"void *\"},child={name=\"var92.5.xHandle.xTasksWaitingToSend.xListEnd.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pvContainer\",exp=\"pvContainer\",numchild=\"3\",value=\"\",type=\"struct xLIST *\"}],has_more=\"0\"\n"},"seq":5058}
1: (13831) ->1414^done,numchild="5",children=[child={name="var92.5.xHandle.xTasksWaitingToSend.xListEnd.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.xItemValue",exp="xItemValue",numchild="0",value="",type="TickType_t"},child={name="var92.5.xHandle.xTasksWaitingToSend.xListEnd.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxNext",exp="pxNext",numchild="5",value="",type="struct xLIST_ITEM *"},child={name="var92.5.xHandle.xTasksWaitingToSend.xListEnd.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious",exp="pxPrevious",numchild="5",value="",type="struct xLIST_ITEM *"},child={name="var92.5.xHandle.xTasksWaitingToSend.xListEnd.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pvOwner",exp="pvOwner",numchild="0",value="",type="void *"},child={name="var92.5.xHandle.xTasksWaitingToSend.xListEnd.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pxPrevious.pvContainer",exp="pvContainer",numchild="3",value="",type="struct xLIST *"}],has_more="0"
--> E (output): {"type":"event","event":"output","body":{"category":"console","output":"1: (13831) ->(gdb)\n"},"seq":5060}
1: (13831) ->(gdb)
--> E (output): {"type":"event","event":"output","body":{"category":"console","output":"1: (13831) 1414: elapsed time 0\n"},"seq":5062}
1: (13831) 1414: elapsed time 0
--> E (output): {"type":"event","event":"output","body":{"category":"console","output":"[Error] 1: (13831) natvis Expand: Object reference not set to an instance of an object.\n"},"seq":5064}
[Error] 1: (13831) natvis Expand: Object reference not set to an instance of an object.

haneefdm, 2025-09-01:
Sorry and thank you for submitting this issue. @malsyned, do you have time to take a look. It is urgent. My own testing was fine but see if you can spot fomething.

@156898293 Details missing. What Debugger? What version of FreeRTOS? etc. Based on the output, I am assuming Cortex-Debug?

haneefdm, 2025-09-01:
Yes gdb. But what debugger front-end are you using in VSCode? ARM, Cortex-Debug, Microsoft? Which extension? We don't talk to gdb at all. We talk to the debugger in the GUI.

156898293, 2025-09-01:
for launch target I set 
"type": "cppdbg",
If I understand your question correctly - it should be: "Microsoft C/C++ extension (ms-vscode.cpptools)"

haneefdm, 2025-09-01:
Yes, cppdbg is "Microsoft C/C++ extension (ms-vscode.cpptools)"

That is not something I tested with. But it helps to know that it was not Cortex-Debug while we investigate

haneefdm, 2025-09-01:
Today is a holiday for me, middle of cooking. I have to investigate first. Bulk of the changes were done by @malsyned 

It would be great if you were available when I am ready with my changes. I have no idea right now what those changes might be. Especially important since I can't duplicate what you are seeing

156898293, 2025-09-01:
ok, I'll be here if you need to debug.

hwmland, 2025-09-01:
I was developing on FreeRTOS v10, now I don't have even this - still no time to setup it after disk issue. 
Would be worth anyway to have information what MCU do you use. There were reports in my PR about problems with ESP32 (version was not standard at that time) And question is what get changed in FreeRTOS v10->V11.
Just to be sure - did you try it as well with some very minimal example, something like FreeRTOS Hello Worls?

156898293, 2025-09-02:
I'm running NXP's i.mx95 - specifically M7 core.

haneefdm, 2025-09-02:
@156898293 When did this happen? When you paused the OS randomly or at a breakpoint in your code?

In the previous version, we did not report on Queues/Semaphores. This is new for the current version but the loop seems to be in the reporting/detection of the Queues/Semaphores.

haneefdm, 2025-09-02:
@malsyned  Yes, guardrails may be necessary.

haneefdm, 2025-09-02:
Sure, a huge chain by itself does not mean it is a runaway loop. But I thought that is what the user is reporting. Lets confirm with @156898293 that it is indeed a runaway loop and when it actually happens.

Also, there may be a way to create new GDB variables for each iteration, but that may actually make it harder to read/follow. We are not expecting hundreds or items here are we?

malsyned, 2025-09-02:
The ultimate error message,
> natvis Expand: Object reference not set to an instance of an object.

is logging a .NET NullReferenceException, so I believe it indicates a bug in MIEngine.

```csharp
        internal IVariableInformation[] Expand(IVariableInformation variable)
        {
            try
            {
                // ... actual function body elided ...
            }
            catch (Exception e)
            {
                _process.Logger.WriteLine(LogLevel.Error, "natvis Expand: " + e.Message);    // TODO: add telemetry
                return variable.Children;
            }
        }
```

This may explain why you are seeing the error with cppdbg but not with cortex-debug. Cortex-debug has its own MI2 protocol implementation and doesn't use MIEngine, right?

malsyned, 2025-09-02:
That's pretty much the end of both my expertise and my time to look into this at the moment. Hope it helps track down the issue.

malsyned, 2025-09-02:
@156898293 can you clarify whether when you said
>  The problem seems to be in infinite loop via gdb:

it was because VSCode was locking up or looping forever printing that error, or if it was because you saw a bunch of linked list pointer dereferences chained together and made an assumption? (it's the same assumption I made at first, is why I ask).

The number of dereferences is still a little suspicious either way, though. Does your project actually have 25 tasks that could all be waiting to send or receive on a single queue?

156898293, 2025-09-03:
Hi, I can reproduce this issue with 100% success rate. I can break before rtos scheduler, I can break in first task, ISR, etc. XRTOS Window is displaying busy updating... and debug window is running out with the messages of this poste. Not sure how long I waited, but it was definitely more than a few minutes.

156898293, 2025-09-03:
I verified on empty hello world application (single task with single printf). The problem is the same in version 0.0.10, while version 0.0.7 seems to be working no problem.

156898293, 2025-09-03:
I also tried to create launch config using cortex-debug, but that seems to be colliding with my lauterbach(t32marm) setup and I can't connect to GDB server (unknown monitor command).

156898293, 2025-09-03:
Again, if you need me to generate any logs, or share .elf/code I can do that for the "hello world" app. We can also arrange debug session. I believe its simple null exception somewhere.

156898293, 2025-09-03:
[xrtos_activation.txt](https://github.com/user-attachments/files/22114904/xrtos_activation.txt)

haneefdm, 2025-09-03:
> I also tried to create launch config using cortex-debug, but that seems to be colliding with my lauterbach(t32marm) setup and I can't connect to GDB server (unknown monitor command).

Maybe we can help you with this. It will be important to rule out cppdbg as being the issue. Can you show us your launch.json for both cppdbg and cortex-debug? You can almost always do what you need with either. The mechanisms may be different. I am assuming you used the `external` server type for cortex-debug.

The Natvis errors are disturbing. This indicates a (potential) bug in cppdbg itself. Because the pointer being dereferenced is a bad CSharp object inside MIEngine and not from this extension or gdb or something we did. I don't want to debug that but it is highly unusual.

> Again, if you need me to generate any logs, or share .elf/code I can do that for the "hello world" app. We can also arrange debug session. I believe its simple null exception somewhere.

Lets rule out cppdbg as the issue first. Then, we can use a shared debug session via Google Meet or something

156898293, 2025-09-03:
I think the lauterbach server doesn't support all the gdb commands. Also my setup is not standard. For instance the unknown "monitor reset halt" (the error) shout reset the target and pause it immediately after.  But in my setup that is quite complicated procedure that I do as a prebuild action and if the reset command really resets the core, my SoC will go into hard-fault.

	{
                "name": "Lauterbach gdb (cppdbg) - Full Init",
                "type": "cppdbg",
                "request": "launch",
                "program": "${command:cmake.launchTargetPath}",
                "targetArchitecture": "arm",
                "cwd": "${workspaceFolder:MyDir}",
                "logging": {
                    "engineLogging": true,
                    "traceResponse": true
                },
                "useExtendedRemote": true,
                "MIMode": "gdb",
                "miDebuggerPath": "${workspaceFolder:ArmToolChain}/bin/arm-none-eabi-gdb",
                "miDebuggerServerAddress": "${config:Lauterbach.addr}:30000",
                "preLaunchTask": "Start GDB server",
                "postDebugTask": "Stop GDB"
            },
            {
                "name": "Lauterbach gdb (cortex-debug) - Full Init",
                "type": "cortex-debug",
                "servertype": "external",
                "request": "launch",
                "executable": "${command:cmake.launchTargetPath}",
                "cwd": "${workspaceFolder:MyDir}",
                "gdbTarget": "${config:Lauterbach.addr}:30000",
                "gdbPath": "${workspaceFolder:ArmToolChain}/bin/arm-none-eabi-gdb",
                "preLaunchTask": "Start GDB server",
                "postDebugTask": "Stop GDB",
                //"device": "YourDeviceName", // Optional, helps with SVD loading
                //"svdFile": "${workspaceFolder}/svd/your_device.svd", // Optional
                "showDevDebugOutput": "both",
                "runToEntryPoint": "main",
                "preLaunchCommands": [],
                "postLaunchCommands": [],
                // "overrideLaunchCommands": [
                //     "${config:Lauterbach.addr}:30000"
                // ],
            },

156898293, 2025-09-03:
> > I think the lauterbach server doesn't support all the gdb commands. Also my setup is not standard. For instance the unknown "monitor reset halt" (the error) should reset the target and pause it immediately after.
> 
> And you don't need to. You can configure what Cortex-Debug uses for connecting/resetting. With `external` server type by default you get commands similar to what openocd needs. You are expected to provide your own. You did almost everything right.
> 
> This page can be helpful in Cortex-Debug Wiki
> 
> https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood#collect-gdbgdb-server-commands-for-start
> 
> The key here is to have a proper `overrideLaunchCommands` and replace it with the commands needed to pause the debugger (if needed and most gdb-servers do) and optionally program the device. You will need at least an empty `overrideLaunchCommands` or else the default commands will be used. Try
> 
>              "showDevDebugOutput": "raw"      // "raw" is more readable than "both"
>             "overrideLaunchCommands": [],
> Once the debug session starts, cortex-debug does not use any more `monitor` commands. It relies on pure gdb commands.

Ok, you are correct. the empty override allowed me to connect. Unfortunately the result is the same error.

[xrtos_activation_cortex_debug.txt](https://github.com/user-attachments/files/22116861/xrtos_activation_cortex_debug.txt)

malsyned, 2025-09-03:
I got curious and decided to play with combining rtos-views and cppdbg this morning. Some observations about that and about the additional details provided by @156898293:

* With the Cortex-R4 project that I've been working on, I can't reproduce this bug with either cortex-debug or cppdbg. In both cases, my XRTOS tab updates normally. This is on Linux (Manjaro), GDB 16.3, cpptools 1.26.3, rtos-views 0.0.10
* The `value=` fields coming back from the GDB server contain values in xrtos_activation.txt, so I think that rules out any relationship to my [previous observation](https://github.com/mcu-debug/rtos-views/issues/71#issuecomment-3246243620) that many of the values were coming back `""` in the original trace. (still weird, though)
* In xrtos_activation.txt, which was created from a small hello world example, I don't see those big chains of `pxPrevious)->pxPrevious`... so I think we can rule out any too-deep-recursion type causes.
* @156898293 can you post your `tasks.json`? I'm curious to see how "Start GDB server" and "Stop GDB" are configured.
* Can you say some more about your platform, software and extension versions? Windows, Linux, or MacOS? What version of the C/C++ extension? What version of GDB? What version of any Lauterbach software?

156898293, 2025-09-04:
Hi,
>In xrtos_activation.txt, which was created from a small hello world example, I don't see those big chains of pxPrevious)->pxPrevious... so I think we can rule out any too-deep-recursion type causes.

I forcibly terminated the debug as my terminal have limited history (i din't want to lose the init sequence). But even the hello world didn't update after 10min.

> can you post your tasks.json? I'm curious to see how "Start GDB server" and "Stop GDB" are configured.

Unfortunately not. Those tasks are running Lauterbach scripts dealing with our CPU, without NDA I can't share them. However those scripts only setup the CPU (GDB server) for running. There is no "memory remapping" or something like this. The .elf debug information is not affected.

>Can you say some more about your platform, software and extension versions? Windows, Linux, or MacOS? What version of the >C/C++ extension? What version of GDB? What version of any Lauterbach software?

Now that you mention it, I realized one thing that may affect this. I'm running vscode on windows (ver. 11-24H2-26100.4946) and connect via "remote-SSH" (ver 0.120.0) to Ubuntu (ver 22.04.5 LTS), where all the other apps are running.
Lauterbach t32marm -    Software Version: N.2025.05.000180334
vscode - Version: 1.103.2 (user setup)
ms-vscode.cpptools - Version 1.26.3
marus25.cortex-debug - Version 1.12.1
mcu-debug.rtos-views - 0.0.7 (working) ; 0.0.10 (not updating)

After morning meetings I'll try to run the vscode directly on ubuntu without windows.

haneefdm, 2025-09-04:
There were at least two positive outcomes that resulted from this issue.
1. Ended up testing the getting of registers properly when done as a tree instead of flat list
2. We found a user with a Lauterbach probe who figured out how to use cppdbg and cortex-debug

156898293, 2025-09-04:
So I checked direct run on target machine and the result is the same (no update). 

One other specific thing came to me - I use only static allocation for everything. Dynamic is disabled in FreeRTOSConfig.h.

[FreeRTOSConfig_Gen.txt](https://github.com/user-attachments/files/22135990/FreeRTOSConfig_Gen.txt)

haneefdm, 2025-09-04:
Static/Dynamic memory allocation is not relevant. Why are the OS data structures (circular list) so long. Because it is a list, we have to read one node at a time. 18K nodes for an RTOS is quite big. Can you explain that @156898293 . You said same happens in a simple hello world. Can you look in the debugger to see why your data is growing so big?

156898293, 2025-09-04:
considering the size of the code I don't think I have something like 18k of struct in there.
m_interrupts:        1612 B         2 KB     78.71%
m_text:       25324 B       254 KB      9.74%
m_m33_suspend_ram:           0 B         8 KB      0.00%
m_a55_suspend_ram:           0 B         8 KB      0.00%
m_data:       14472 B       104 KB     13.59%
m_ncache:           0 B        64 KB      0.00%
m_reserved:           0 B        56 KB      0.00%
m_rsc_tbl:           0 B        16 KB      0.00%
m_rsc_tbl_in_dram:           0 B        16 KB      0.00%

156898293, 2025-09-05:
Please note that I forbid dynamic allocation.
Looking at xQueueRegistry table:
/*
 * The queue registry is just a means for kernel aware debuggers to locate
 * queue structures.  It has no other purpose so is an optional component.
 */
#if ( configQUEUE_REGISTRY_SIZE > 0 )
      PRIVILEGED_DATA QueueRegistryItem_t xQueueRegistry[ configQUEUE_REGISTRY_SIZE ];
#endif
my FreeRTOSConfig.h:
#define configQUEUE_REGISTRY_SIZE 8

156898293, 2025-09-05:
[image]

156898293, 2025-09-05:
Wow! Great. I even got my runtime profiling stats.

...Yes, the fix seem to be working.

haneefdm, 2025-09-05:
Thank you for reporting this, and thanks to the people who jumped in to help.